Advanced3dPrinting/A3dp-Slicer #2: Pressure Advance per filament not outputting to file correctly

lukeslaboratory, 2022-12-24:
### Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. [Development builds of Slic3r](https://dl.slic3r.org/dev). [Prusa3D Issue Tracker](https://github.com/Prusa3D/Slic3r/issues)

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**3D Model and Slic3r Configuration Export**
Please upload a ZIP archive containing a copy of the 3D model you are seeing this bug with and a CTRL-E export of the configuration used.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. MacOS, Linux, Windows]
 - Version [e.g. 1.3.0]
     - _Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._
     - _For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
     - _Use `About->About Slic3r` for release versions._

**Additional context**
Add any other context about the problem here.

lukeslaboratory, 2022-12-24:
Also - idea for this feature:
Allow autoscaling for PA - essentially give 2-3 PA values, at 2-3 matching acceleration values, (so for instance at 30k accel its a pa of .01, at 15k its pa of .02, at 25k its .013, etc) so you can just autoscale between them depending on the features acceleration.

robthide37, 2022-12-25:
Fixed

closed 2023-09-03 by robthide37