Advanced3dPrinting/A3dp-Slicer #5: Multiple Configuration Folder Option

jallen810, 2023-03-07:
We need ability to have multiple configuration folders for different machines that are completely, mutually independent. 
Ideally the user could have a drop-down in-app to load this different configuration folder presets and there is no way of cross contamination.

robthide37, 2023-07-20:
You can make a folder named configuration in the main directory instead of using the shortcut method i previously showed

jallen810, 2023-07-22:
This only partially solves my problem specifically. I would ideally like to map my profiles to a cloud drive location (aka NextCloud/OneDrive/etc..) which are 2-way synced to my local machine C-drive on multiple devices and would like to have the following folder structure to keep my printers separated cleanly.

                       Cloud Drive Config folder directory (f.e. C:/OneDrive/A3DPSlicerConfigurationFolder)
                                            |
                                             --- Printer 1 Folder
                                                       -Bed Shape / Texture
                                                       - Print/Filament/Printer Profiles
                                            |
                                            |  --- Printer 2 Folder
                                                       -Bed Shape / Texture
                                                       - Print/Filament/Printer Profiles         
                               
                                            |
                                            |  --- Printer 3 Folder
                                                       -Bed Shape / Texture
                                                       - Print/Filament/Printer Profiles