Advanced3dPrinting/A3dp-Slicer #6: Allow A3DP to import in Solid model CAD files (.STP/.STEP/.F3D/.IPT/.sldprt/etc)

jallen810, 2023-03-09:
I have heard of some of the other slicers working on allowing us to import .STP/.STEP/.f3d/.ipt/.sldprt/etc files directly from CAD programs so we don't have to settle for the losses that come from .stl files.

I realize there are many challenges for some of these file types (licensing being the major thing) but .STP/.STEP should be feasible...

robthide37, 2023-07-20:
This is done

closed 2023-07-20 by robthide37

jallen810, 2023-07-22:
Rob,

 

Can you also see if you can get Fusion360 & Inventor file types to load in natively? 

Those would be the following:

 

Fusion 360

.f3d 

.f3z

 

Inventor

.ipt

.iam 

 

Best,

Jake

 

Closed #6 <https://github.com/robthide37/A3dp-Slicer/issues/6>  as completed.